MatthewKuKanich/FindMyFlipper #1: Error running generate_keys.py

nortakales, 2024-03-08:
I'm getting an error when running `python3 generate_keys.py`. I'm sure I'm just doing something wrong, or maybe this only works on a newer version of python?

Python 3.8.10
Ubuntu 20.04.6 LTS
WSL2

```
Traceback (most recent call last):
  File "generate_keys.py", line 86, in <module>
    main()
  File "generate_keys.py", line 59, in main
    public_key_hash = hashes.Hash(hashes.SHA256())
TypeError: __init__() missing 1 required positional argument: 'backend'
```

Luk164, 2024-03-08:
Weird, the script worked fine for me after I installed cryptography.
[image]

nortakales, 2024-03-08:
```
$ pip3 install cryptography
Requirement already satisfied: cryptography in /usr/lib/python3/dist-packages (2.8)
```
Seems it was already installed for me. I must have an incorrect or outdated version of something (just my guess), but I can't find any mention of minimum version required.

MatthewKuKanich, 2024-03-08:
> ```
> $ pip3 install cryptography
> Requirement already satisfied: cryptography in /usr/lib/python3/dist-packages (2.8)
> $ pip3 check cryptography  
> No broken requirements found.
> ```
> 
> Seems it was already installed for me. I must have an incorrect or outdated version of something (just my guess), but I can't find any mention of minimum version required.

You may need to update your python or try in a separate instance/installation of python.

nortakales, 2024-03-08:
Found another machine with Ubuntu 22 and python 3.10 which worked. Seems that must be the minimum requirement! Now we know.

closed 2024-03-08 by nortakales

FatherDivine, 2024-03-08:
> Found another machine with Ubuntu 22 and python 3.10 which worked. Seems that must be the minimum requirement! Now we know.

The latest version of Python is 3.12.2. You are still behind quite a bit, as 3.7 and below are not unsupported anymore. If you don't have a reason for running an old version (which introduces security risks), I suggest you update to the latest version of python.