MatthewKuKanich/FindMyFlipper #3: Precompiled "faps" and building only works on dev branches of flipper firmware

ihrapsa, 2024-03-08:
If you try to build the fap it will cry for `extra_beacon.h`. I noticed that's only available in the `dev` branch `targets`. The precompiled `faps` give a similar error when running on `release` firmware: `Missing Imports`

Did not find this documented and I thought it would be useful if anyone else stumbles upon these issues.

Thanks for putting this together!

MatthewKuKanich, 2024-03-08:
Thanks for reminding me, I added it to the instructions for other to see. Hopefully, release will receive the extra BLE beacon api soon. I'm leaving this open for anyone who runs into this to see

YupengLai4, 2024-03-09:
Hi! I'm still getting the `Missing Imports` error, do you think it could be because I upgraded straight from stable to dev?

[image]

jacksonattwood, 2024-03-09:
That shouldn't be an issue. Try compiling the dev branch from CLI:
```
git clone  https://github.com/Flipper-XFW/Xtreme-Firmware.git --branch dev
cd Xtreme-Firmware/
./fbt flash_usb_full
```

MatthewKuKanich, 2024-03-09:
> Hi! I'm still getting the `Missing Imports` error, do you think it could be because I upgraded straight from stable to dev?
> 
> ![screenshot of the error](https://private-user-images.githubusercontent.com/57296300/311397329-24bb66d7-31fa-4f66-ac21-f17c47e176bf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5NDQ2MjcsIm5iZiI6MTcwOTk0NDMyNywicGF0aCI6Ii81NzI5NjMwMC8zMTEzOTczMjktMjRiYjY2ZDctMzFmYS00ZjY2LWFjMjEtZjE3YzQ3ZTE3NmJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA5VDAwMzIwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyMjM1OTJlMGZmZDczM2NiMGNiZGE0YTVhYjZjYWM0MmNmYjYyNDVhODgyYTdlNGUwYjk1YjFhNTAzOTUyZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ebkeF4L8TknS9zAniGFkDxvi-1_Hkm6SzmnnkKJwAy0)

First you need to update your flipper to the dev firmware. Delete the current .fap and replace it with the latest Xtreme fap in the Releases folder. Let me know if it still throws the error

YupengLai4, 2024-03-09:
> ```
> git clone  https://github.com/Flipper-XFW/Xtreme-Firmware.git --branch dev
> cd Xtreme-Firmware/
> ./fbt flash_usb_full
> ```
this worked! Thanks so much for the prompt response :)

Awlexegrecki, 2024-03-09:
> ```
> git clone  https://github.com/Flipper-XFW/Xtreme-Firmware.git --branch dev
> cd Xtreme-Firmware/
> ./fbt flash_usb_full
> ```
I got this error:
```
 File "C:\Users\<user>\Downloads\Xtreme-Firmware\applications\external\air_arkanoid\engine\scripts\sprite_builder.py", line 8, in <module>
    from PIL import Image, ImageOps
ModuleNotFoundError: No module named 'PIL'
scons: *** [applications\external\air_arkanoid\assets] Error 1

********** FBT ERRORS **********
applications\external\air_arkanoid\assets: Error 1
```

jacksonattwood, 2024-03-09:
> ```
>  File "C:\Users\<user>\Downloads\Xtreme-Firmware\applications\external\air_arkanoid\engine\scripts\sprite_builder.py", line 8, in <module>
>     from PIL import Image, ImageOps [...]

Interesting. Can you try deleting the ```"Xtreme-Firmware"``` folder, and repeat the commands.

Awlexegrecki, 2024-03-09:
I received the same error @jacksonattwood

MatthewKuKanich, 2024-03-09:
> I received the same error @jacksonattwood . Should I try to download firmware from Unleashed?

It should work if you re-clone Xtreme, then add the findmy source to Applications_user and build. Delete the firmware folder you have now to be safe

nortakales, 2024-03-09:
Sort of related based on the issue title (seems like this *should* build against dev branch): I am building with `fbt fap_dist` against the dev branch of the official firmware and getting:

```
applications_user/FindMyFlipper/scenes/findmy_scene_config_mac.c:20:5: error: implicit declaration of function 'furi_hal_bt_reverse_mac_addr' [-Werror=implicit-function-declaration]
   20 |     furi_hal_bt_reverse_mac_addr(app->mac_buf);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

MatthewKuKanich, 2024-03-09:
> Sort of related based on the issue title (seems like this _should_ build against dev branch): I am building with `fbt fap_dist` against the dev branch of the official firmware and getting:
>
> ``` [...]

Which firmware are you building for?

nortakales, 2024-03-09:
> > Sort of related based on the issue title (seems like this _should_ build against dev branch): I am building with `fbt fap_dist` against the dev branch of the official firmware and getting:
> > ```
> > applications_user/FindMyFlipper/scenes/findmy_scene_config_mac.c:20:5: error: implicit declaration of function 'furi_hal_bt_reverse_mac_addr' [-Werror=implicit-function-declaration]
> >    20 |     furi_hal_bt_reverse_mac_addr(app->mac_buf);
> >       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> > ```
> 
> Which firmware are you building for? This function is not available in OFW but is available in the other custom firmware. For OFW I wrote my own reverse mac function for the OFW build, its in the OFW branch so use that if compiling for OFW

I was building for OFW, didn't realize you had an OFW branch. That worked!

ihrapsa, 2024-03-10:
This is weird. 
I'm on `Unleashed` fw on latest dev build. However, the latest precompiled app does not work. I also tried compiling it myself but it's giving me the same `Missing Imports` error when launching it. If I switch to the previous precompiled app it works as usual.

```
2024-03-10 16:49:01,158 [ERROR] Unexpected response: Load failed, /ext/apps/Bluetooth/findmy.fap: Update Firmware to use with this Application (MissingImports)
scons: *** [build/f7-firmware-C/site_scons/phony_launch] Error 3

********** FBT ERRORS **********
build/f7-firmware-C/site_scons/phony_launch: Error 3

```

Any idea what's going on?
Thanks!

nortakales, 2024-03-14:
Confirmed working on OFW release branch as of 0.99.1

closed 2024-03-23 by MatthewKuKanich