MatthewKuKanich/FindMyFlipper #5: No module named cryptography

Awlexegrecki, 2024-03-09:
File "C:\Users\<user>\Downloads\generate_keys.py", line 3, in <module>
    from cryptography.hazmat.primitives.asymmetric import ec
ModuleNotFoundError: No module named 'cryptography'

This was shown when I tried running generate_keys.py

MatthewKuKanich, 2024-03-09:
You'll need to install the cryptography package/module. In terminal type "python3 -m pip install cryptography"

Awlexegrecki, 2024-03-09:
thanks let me try that

MatthewKuKanich, 2024-03-09:
Some python installations are weird, I'm using python 3 but I have to use ```python -m pip install``` instead of ```python3 -m pip install``` So if it doesn't work right away try that

closed 2024-03-09 by Awlexegrecki

Awlexegrecki, 2024-03-26:
I just abandoned the idea of FindMy flipper for myself

MatthewKuKanich, 2024-03-26:
@SgtBlade @Awlexegrecki First make sure you've followed the updated instructions and did this in a python virtual environment. If it still doesn't work, then you can try this in the Windows subsystem for Linux. If that fails, then you can try on a different machine or a Linux machine. If you want to avoid this headache entirely then do the cloning method. It is SUPER easy and lets you use tracking apps on your phone to location your Flipper. You can even use knock off trackers and they work just as well ($2 or so)

SgtBlade, 2024-03-27:
I have found one part of the problem. Even though I work in the venv, all packages get downloaded to global. When trying to execute the script it looks for the packages in the venv. The (bad) fix for this is to just copy and paste the packages from global to this venv. 
This allowed me to run the generate_keys.py and request_reports.py script. I'm still getting an error on the  RequestReport&Map.py script for which I still have to look why and how to fix it

I have no idea why it's doing this or how to fix it put my solution worked.

MatthewKuKanich, 2024-03-28:
> I have found one part of the problem. Even though I work in the venv, all packages get downloaded to global. When trying to execute the script it looks for the packages in the venv. The (bad) fix [...]

Thank you for your insight! Did you install the modules after activating the venv or before? This will change the install. I also have a requirements.txt that should automate it and avoid the error. Could you open a new issue with the error you're getting with the RequestReport&Map.py script? Thanks!