MatthewKuKanich/FindMyFlipper #6: FAP does not work with Flipper Unleashed-071

Awlexegrecki, 2024-03-09:
Tried running, said app was out of date

jacksonattwood, 2024-03-09:
Have a look at #3 and see if that helps

ldxjy, 2024-03-09:
yeah, same issue, there is no dev build unleashed firmware.

UberGuidoZ, 2024-03-09:
That is not true. There may not be a pre-compiled release, but you can follow the same steps as above, substituting the DarkFlippers git in place of Xtreme. (I’m on mobile or I’d give you the exact command.)

jacksonattwood, 2024-03-09:
> yeah, same issue, there is no dev build unleashed firmware. 

You can compile the latest dev build yourself (it seems to be the default branch on the repo) 

```
git clone  https://github.com/DarkFlippers/unleashed-firmware.git --branch dev
cd unleashed-firmware/
./fbt flash_usb_full
```

At time of writing this, there has been 218 commits to the dev branch since the last release

ldxjy, 2024-03-09:
> You can compile the latest dev build yourself (it seems to be the default branch on the repo)
>
> ```
> git clone  https://github.com/DarkFlippers/unleashed-firmware.git --branch dev
> cd [...]

Ok thanks!

ldxjy, 2024-03-09:
> That is not true. There may not be a pre-compiled release, but you can follow the same steps as above, substituting the DarkFlippers git in place of Xtreme. (I’m on mobile or I’d give you the exact command.)

I see, thanks !

closed 2024-03-23 by MatthewKuKanich