MatthewKuKanich/FindMyFlipper #7: Importing generated keys in openhaystack

lupettohf, 2024-03-09:
I'm trying to import the generated keys inside OHS however the import function looks for a .plist file; I tried exporting one and it seems holding more keys than the one generated by the python script. Any additional step to perform?

Martyfree123, 2024-03-10:
Having the same issue. Have not used OpenHaystack before so not sure how to proceed. @MatthewKuKanich if you would be so kind as to add a section to the readme documenting importing into OpenHaystack and adding to the iOS FindMy app I would be really grateful!

Chapoly1305, 2024-03-10:
> Having the same issue. Have not used OpenHaystack before so not sure how to proceed. @MatthewKuKanich if you would be so kind as to add a section to the readme documenting importing into OpenHaystack and adding to the iOS FindMy app I would be really grateful!

Dear @Martyfree123 ,
I am happy to assist in questions and make clarification. As you asked, 
1. To use OpenHayStack, you must use MacOS lower than version 14 (https://github.com/seemoo-lab/openhaystack/discussions/224). If you do not own a Mac device or the system has been upgraded to 14 and beyond. The alternative solution includes, 

- https://github.com/dchristl/macless-haystack (recommended in README)
- https://github.com/Chapoly1305/FindMy (a project uses python and docker to provide location lookup as a backend service)
2. Unfortunately, this project and OpenHayStack are both unlikely able to add into the Apple FindMy app, due to the device in such compatibility requires to pay and pass the Apple MFi certification. You may consider use software mentioned, e.g., [OwnTags](https://owntracks.org). With the combination of https://github.com/Chapoly1305/FindMy, you can have a similar result.

closed 2024-03-23 by MatthewKuKanich