MatthewKuKanich/FindMyFlipper #8: Payload input field lenght does not match generated payload

Programmer043, 2024-03-09:
Generating payload with generate_keys.py gives me 50 character long (25 HEX pairs) payload, but the app is asking for payload 62 characters long. Do I enter the MAC there or leave it 0s?

ihrapsa, 2024-03-09:
First 6 bytes are static and already inserted for you. You should enter the generated payload only after that.

nortakales, 2024-03-09:
I noticed this too. First 7 bytes of my payload from generate_keys.py are the same as the first 7 bytes of the default value in the app: `1E FF 4C 00 12 19 00` so I left those alone. I entered the rest of the bytes which actually differed and came up 7 bytes short, which I left as all `00`. Are we supposed to repeat `1E FF 4C 00 12 19 00` twice at the beginning then?

MatthewKuKanich, 2024-03-09:
> I noticed this too. First 7 bytes of my payload from generate_keys.py are the same as the first 7 bytes of the default value in the app: `1E FF 4C 00 12 19 00` so I left those alone. I entered the rest of the bytes which actually differed and came up 7 bytes short, which I left as all `00`. Are we supposed to repeat `1E FF 4C 00 12 19 00` twice at the beginning then?
> 
> I've gone through all the steps to get this working with OpenHaystack and it isn't working still, this is the only thing I can think of so far that might be causing it not to work.

Hi, I'm taking a look now. The first 7 bytes should **Not** repeat, These are static bytes that work as identifiers. I'm checking the script now and I'll get back shortly

MatthewKuKanich, 2024-03-09:
@Programmer043 @nortakales @ihrapsa Pull changes from latest push, the python script has been fixed and you can now put the file it generates on your Flipper SD card and import the keys instead of having to type it in manually

Martyfree123, 2024-03-10:
Having the same issue. I am new to flipper so I have not used ufbt before. Are you in the process of creating new precompiled .fap packages?

MatthewKuKanich, 2024-03-10:
> Having the same issue. I am new to flipper so I have not used ufbt before. Are you in the process of creating new precompiled .fap packages?

I just recompiled faps for several firmware, check releases folder. I recommend you learn how to use ufbt though, as updates will come at a point where I won't be able to precompile faps for all firmware in a timely manner.

nortakales, 2024-03-10:
On the latest OFW version of FindMy, I am noticing two things related to this issue:

1. When manually entering payload, there are still more bytes in the input field than there are in my .keys file
2. When importing a .keys file, and later inspecting `findmy_state.txt` on the SD card, the MAC and data are not the same as in my .keys file. Example below.

1RnlRPz.keys
```
Private key: ZPyasvqMnnKNssqomku3bOgd/4NBuTVoFIQklA==
Public key: OqNPSmlDZegpB9QkJRnM5W5hYHpNLY6JyzN7Mg==
Hashed adv key: 1RnlRPzz2VpM20hoqm3WP6qP4h4fF8WQ3hdyhFwtoIo=
Private key (Hex): 64fc9ab2fa8c9e728db2caa89a4bb76ce81dff8341b9356814842494
Public key (Hex): 3aa34f4a694365e82907d4242519cce56e61607a4d2d8e89cb337b32
MAC: 3aa34f4a6943
Payload: 1eff4c0012190065e82907d4242519cce56e61607a4d2d0000
```

findmy_state.txt
```
Filetype: FindMy Flipper State
Version: 1
beacon_active: false
broadcast_interval: 1
transmit_power: 6
mac: FA A3 4F 4A 69 43
data: 1E FF 4C 00 12 19 00 65 E8 29 07 D4 24 25 19 CC E5 6E 61 60 7A 4D 2D 8E 89 CB 33 7B 32 00 00
```
Differences:
* MAC differs just in the first character, `3` -> `F`
* data has `8E 89 CB 33 7B 32` inserted before the final `00 00`

I still have not been able to get this app working with OpenHaystack and I suspect maybe this is why. I just ordered an AirTag (should be here this afternoon) to give cloning that a try, but if there are general issues with just getting the correct MAC/payload into the app I think neither method would be working.

nortakales, 2024-03-10:
I updated my script, generated a new .keys file, imported it into the app, and inspected findmy_state.txt. Everything matches perfectly now! Thanks!

Still can't get it working with OpenHaystack (Mac-less) though. However, nothing points to an issue with this app now, it is probably something else with my setup. 

Like I said, I'll give cloning a real Air Tag a try this afternoon when it gets delivered. Would be nice to have it show directly in the Find My iOS app (I assume that is how it work, right?), but obviously that comes at a slight cost.

MatthewKuKanich, 2024-03-10:
> I updated my script, generated a new .keys file, imported it into the app, and inspected findmy_state.txt. Everything matches perfectly now! Thanks!
> 
> Still can't get it working with OpenHaystack (Mac-less) though. However, nothing points to an issue with this app now, it is probably something else with my setup.
> 
> Like I said, I'll give cloning a real Air Tag a try this afternoon when it gets delivered. Would be nice to have it show directly in the Find My iOS app (I assume that is how it would work, right?), but obviously that comes at a slight cost.

Macless haystack is a bit of a pain to setup, you can try the second macless method by following the other link in the Readme. That may be easier to get going. As for cloning tags, you first register them in the FindMy iOS app and it will continue to update there once cloned to your Flipper which is a huge quality of life advantage over generating your own. I'm looking into knock off airtags that work with the official findmy app to see if I can reverse engineer their method of registration.

MatthewKuKanich, 2024-03-11:
> Cloned my air tag and it works! One thing to note: I tried to take a short cut and just modify `findmy_state.txt` directly since it was ever so slightly easier than importing another .keys file updated with my Air Tag's MAC and payload. It was not working. I went into the Find My app on the Flipper > Register New Tag and observed that the MAC was reversed (from what I entered in `findmy_state.txt`). I manually entered the MAC and suddenly my iPhone notified me it found the Air Tag. Super cool!

Thanks for the detailed update! So about the MAC reversal, for some reason the implementation of the extra beacon requires setting the MAC address in reverse order. I'm not sure why it was made this way but it is what it is. To counteract that I added a function that reverses the MAC once it's been input so that the user can input the MAC in the correct order and it will do the reversal for you. This however won't work if you bypass this by manually typing in the data as this function won't run.

closed 2024-03-23 by MatthewKuKanich