MatthewKuKanich/FindMyFlipper #10: ofw branch not building against OFW

nortakales, 2024-03-10:
I'm on the latest commit of the OFW `dev` branch, and the latest commit of the `ofw` branch for FindMyFlipper `7ce772c534715a67cdd9d06419cf3a5c708e4702`, but it is not building. Prior commit `7e6b8f386b8b6a09e4d4e8361f85a230e7a8bfc4` was building.

Error:

```
********** FBT ERRORS **********
build/f7-firmware-D/.extapps/findmy.impsyms: build/f7-firmware-D/.extapps/findmy.fap: app may not be runnable. Symbols not resolved using firmware's API: {'I_Lock_7x8', 'I_Ok_btn_9x9', 'I_WarningDolphinFlip_45x42', 'I_DolphinDone_80x58'}
```

MatthewKuKanich, 2024-03-10:
Hi, yes there are some issues with icons atm. I'm going to remove them and it should build.

MatthewKuKanich, 2024-03-10:
> I'm on the latest commit of the OFW `dev` branch, and the latest commit of the `ofw` branch for FindMyFlipper `7ce772c534715a67cdd9d06419cf3a5c708e4702`, but it is not building. Prior commit [...]

Should be fixed now! Along with fixed precompiled faps

closed 2024-03-10 by MatthewKuKanich

nortakales, 2024-03-10:
Probably not the correct long term fix, but I got it building *with* the icons by copying the relevant icons from the flipper firmware over to `/icons`, renaming them with a prefix (to avoid redeclaration errors), and updating the code references to the icons to also include the prefix.

I am not aware of the correct solution to re-using the existing icons, but I imagine it must be possible.

MatthewKuKanich, 2024-03-10:
> Probably not the correct long term fix, but I got it building _with_ the icons by copying the relevant icons from the flipper firmware over to `/icons`, renaming them with a prefix (to avoid [...]

Yeah I've considered that approach, I likely will end up making that change. I wanted to quickly get a working build up on the repo so thats why I quickly commented out the icons. Thanks for confirming that methods works!