MatthewKuKanich/FindMyFlipper #14: no beacons transmitted

akoskoronka, 2024-03-11:
Hello,

The app shows "Broadcast Active", without any errors anywhere, however no beacons are actually transmitted.
State file shows mac and data correctly.
Using RogueMaster built from latest git, which includes this app.
Any ideas?

Thanks.

MatthewKuKanich, 2024-03-11:
> Hello,
> 
> The app shows "Broadcast Active", without any errors anywhere, however no beacons are actually transmitted. State file shows mac and data correctly. Using RogueMaster built from latest git, which includes this app. Any ideas?
> 
> Thanks.

Have you verified no beacon being sent via nrf Connect app?

akoskoronka, 2024-03-11:
Yes, nrf connect and airguard as well.

MatthewKuKanich, 2024-03-11:
Interesting. Try building with my repo and see. If my repo works I'll talk to rogue master and figure out the issue. This is the first time I've heard of this happening so I want to cover all possibilities

akoskoronka, 2024-03-13:
Very weird, neither nrf or airguard can pick up the signal, but a few location reports are available.
I think, that sometimes it is able to broadcast the ping, but surely not according to the settings.
I am currently using RM0311, broadcast_interval 1 and transmit_power 6, which resulted in most reports.
I would need to figure it out how to list all the location reports from macless-haystack to further check the timestamps and each location reported.
Any other ideas on how to debug the situation?
Thanks!

MatthewKuKanich, 2024-03-23:
> Very weird, neither nrf or airguard can pick up the signal, but a few location reports are available. I think, that sometimes it is able to broadcast the ping, but surely not according to the [...]

Use the latest tutorial (theres a video) it lets you map out and list everything. It is working as expected from what I understand. I would tweak nrfConnect filters as it must be transmitting if you were able to get location reports.

closed 2024-03-23 by MatthewKuKanich