MatthewKuKanich/FindMyFlipper #18: Payload starts with 0x on airtag

kenneth7398, 2024-03-12:
my payload starts with 0x on find my flipper so i dont know how to add it manually as there is no x to type

MatthewKuKanich, 2024-03-12:
Hey! You can ignore the 0x, in Hex/Byte format the numbers normally lead with 0x to show the beginning of a byte or hex value. So instead of 64, it would be 0x64. This helps people know that the format is in hex and not decimal. Since 64 does not equal 0x64. Hope this helps!

FastblinkerCalamity, 2024-03-12:
You don’t type the 0xI’m running it with a cloned AirTag and seems to work fine it gives the position on the iPhone app the only thing is you can’t send sound or use the tracking thing with the arrow.Richard ClarkA.k.a FastBlinker von Calamityhttp://www.instagram.com/exit_stage_left/http://FastBlinker.com/

closed 2024-03-12 by MatthewKuKanich