MatthewKuKanich/FindMyFlipper #24: Samsung mode

einsjakob, 2024-03-14:
Hi, I don't know if it's me, but I can't find a way to switch to the Samsung network. Can someone help me with that?

Marc-Harony, 2024-03-14:
Hi 👋🏻 ,
Have you checked issue #19 ?
I have the same problem, here is the explanation:
> Hey! Yes! I've wrote the app to auto-detect which network to use based on the key you use. So if you type in/import a Samsung key it will switch to the Samsung network automatically.

Marc-Harony, 2024-03-14:
Not sure but I think you may edit the [generate_keys.py](https://github.com/MatthewKuKanich/FindMyFlipper/blob/main/KeyGeneration/generate_keys.py#L12) file.
If you check on line 12 you see an hex number corresponding to apple's company ID:
[image]

Sadly, I have no idea what Samsung's ID might be.

MatthewKuKanich, 2024-03-14:
Hi! Unfortunately you can't generate a Samsung SmartTag key yet. To use the Samsung find network you'll have to clone a legit and registered tag

einsjakob, 2024-03-14:
Ok, thank you for the response. I guess then I will have to wait until its possible to generate them 😅

Calevra1, 2024-03-19:
I got a smart tag but dont now how to clone it

einsjakob, 2024-03-19:
You can look at step two in the README

closed 2024-03-23 by MatthewKuKanich