MatthewKuKanich/FindMyFlipper #25: Character overly in configuration line

gofferman, 2024-03-14:
I just installed this app ( running 0.99.1 of FZ firmware ) and went to the config and the bottom line has characters overlaid.

[image]

Marc-Harony, 2024-03-14:
Hi :wave:,
same problem here.
I'm using the latest Official FW version on dev branch (022fcc0)
It does not appear on Momentum Firmware

MatthewKuKanich, 2024-03-14:
Yikes, thanks for letting me know! Fortunately it'd just the credits that are bugged. I'll get a fix out tonight though

dogtopus, 2024-07-08:
Other than giving credits its own scene, I'd also suggest changing the `Broadcast Interval` and `Transmit Power` to `TX Interval` and `TX Power` respectively to make them look less crowded.