MatthewKuKanich/FindMyFlipper #29: "A server problem is blocking Apple ID sign in" after "request_reports" script execution.

erichamberg, 2024-03-21:
All setup and execution occurs as described in your documentation, but upon successful 2FA auth against Apple, I receive a login error. 

[me@here AirTagGeneration]$ ~/containerData/anisette/venv/bin/python3 request_reports.py
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
2FA required, requesting code
pyprovision is not installed, querying http://localhost:6969 for an anisette server
Enter 2FA code: 416319
2FA successful
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
b'<?xml version="1.0" encoding="UTF-8" standalone="yes"?>\n<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">\n<plist version="1.0">\n\t<dict>\n\t\t<key>dsid</key>\n\t\t<string>xxxxxxxxxxx</string>\n\n\t\t<key>delegates</key>\n\t\t<dict>\n\t\t\t<key>com.apple.mobileme</key>\n\t\t\t<dict>\n\t\t\t\t<key>status</key>\n\t\t\t\t<integer>1</integer>\n\n\t\t\t\t<key>status-message</key>\n\t\t\t\t<string>A server problem is blocking Apple ID sign in. Try signing in later.</string>\n\n\t\t\t</dict>\n\n\t\t</dict>\n\n\t\t<key>status</key>\n\t\t<integer>0</integer>\n\n\t</dict>\n</plist>'
'service-data'

For background, I have NO Apple devices and created the Apple ID for this purpose only. The fake MacBook DOES appear in my "Devices" once this process completes, though. I'm not certain the issue isn't beyond the scope of this app, either. Any input is appreciated, as I'm very inexperienced with Apple's authentication schemes.

AngelRamirezE, 2024-03-22:
I'm getting same error. Then I ran web_service.py and I got this message: 

Traceback (most recent call last):
  File "...\FindMy\AirTagGeneration\web_service.py", line 49, in <module>
    'searchPartyToken': mobileme['delegates']['com.apple.mobileme']['service-data']['tokens'][
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'service-data'

erichamberg, 2024-03-23:
Understood. I did not know an apple product was required.

Thank you!

> I'm getting same error. Then I ran web_service.py and I got this message:
> 
> Traceback (most recent call last): File "...\FindMy\AirTagGeneration\web_service.py", line 49, in 'searchPartyToken': mobileme['delegates']['com.apple.mobileme']['service-data']['tokens'][ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^ KeyError: 'service-data'
> 
> edit: I refereed to this issue: [Chapoly1305/FindMy#9](https://github.com/Chapoly1305/FindMy/issues/9) Seems that we need to log in on an authentic apple product to upgrade the account. I solved that way. Then I got M2 error but because I selected "prefer call for 2fa".

closed 2024-03-23 by erichamberg

MatthewKuKanich, 2024-03-25:
> Understood. I did not know an apple product was required.
>
> Thank you!

There are a couple ways to make an apple account legit. I haven't tested but I've heard that making it then buying a song on iTunes works.