MatthewKuKanich/FindMyFlipper #35: 2FA / Code Error

dillkill210, 2024-03-25:
I downloaded the modules like cryptography etc but i get this error when running request_reports.py:

 line 17, in <module>
    from cores.pypush_gsa_icloud import icloud_login_mobileme, generate_anisette_headers
  File "C:\Users\Desktop\FindMyFlipper\AirTagGeneration\cores\pypush_gsa_icloud.py", line 15, in <module>
    from Crypto.Hash import SHA256
ModuleNotFoundError: No module named 'Crypto'

I've got pycryptodome and Crypto installed and I also used AI to rectify the error and even when it asked for the 2FA i got nothing on my phone, no sms.

Hope you can help

YupengLai4, 2024-03-25:
I got the same problem. This problem should be on the openstack [2FA problem](https://github.com/dchristl/macless-haystack/issues/46).

MatthewKuKanich, 2024-03-25:
> I downloaded the modules like cryptography etc but i get this error when running request_reports.py:
> 
> line 17, in from cores.pypush_gsa_icloud import icloud_login_mobileme, generate_anisette_headers File "C:\Users\Desktop\FindMyFlipper\AirTagGeneration\cores\pypush_gsa_icloud.py", line 15, in from Crypto.Hash import SHA256 ModuleNotFoundError: No module named 'Crypto'
> 
> I've got pycryptodome and Crypto installed and I also used AI to rectify the error and even when it asked for the 2FA i got nothing on my phone, no sms.
> 
> Hope you can help

Try changing the phone number on your apple account for 2fa. A common cause is an outdated phone number

EAJRD, 2024-03-30:
Hello, it won't let me go past 2 FA, I never get the SMS, but I enter the Apple ID ---> login and security --> two-factor authentication --> get a code, everything is fine Apple me It gives a code but I put it in the command line and it asks me for the code again and it keeps asking me until it gives me an error 'M2'

If anyone can give me a hand on this

EAJRD, 2024-03-30:
> If SMS code verification doesn't work (It didn't for me either use the argument -t when executing the request_reports script (or any others as well I believe.). It sends the verification code to a trusted device instead of a phone number

this worked for me