MatthewKuKanich/FindMyFlipper #36: Find my randomly stops tracking

kenneth7398, 2024-03-25:
I’ve had an issue where find my flipper will stop tracking randomly even after it’s set to track in the background. I have to go back in the app for it to work again then it starts tracking again

MatthewKuKanich, 2024-03-25:
How do you know it stops tracking?

kenneth7398, 2024-03-25:
I’m using momentum and I know it stops tracking because I’ve had it happen a few times where it should be active but I check my find my app and it says like 2 hours ago for example.

MatthewKuKanich, 2024-03-25:
> I’m using momentum and I know it stops tracking because I’ve had it happen a few times where it should be active but I check my find my app and it says like 2 hours ago for example. 

Interesting. Does the beacon icon still show on the animations screen? It's a little signal icon that appears once you enter background mode.

Could you also tell me your current settings in the settings app? Specifically the settings in the System tab

kenneth7398, 2024-03-25:
I believe the icon was still showing and it was not tracking through find my. If it happens again I’ll screenshot find my and the flipper icon because when I open the app it fixes it the only problem is if you lose it you won’t be able to open the app. The setting I use is 10s interval and 6dbm.

MatthewKuKanich, 2024-03-26:
> It did it again. I left it active and it randomly turned off. The thing that says if it’s active turned off. 

This definitely seems like a flipper issue. What is your Flippers sleep mode set to?

runestone74, 2024-03-27:
Is it normal to have to restart the app after using Bluetooth for other apps? Can this be fixed with some watchdog process?

MatthewKuKanich, 2024-03-27:
> Is it normal to have to restart the app after using Bluetooth for other apps? Can this be fixed with some watchdog process?

This is not normal behavior. There is a setting in most custom firmware to allow the use of RPC/Bluetooth while a beacon is active.

kenneth7398, 2024-03-27:
> > It did it again. I left it active and it randomly turned off. The thing that says if it’s active turned off.
> 
> This definitely seems like a flipper issue. What is your Flippers sleep mode set to?

My sleep mode is 120s before the screen dims but it is set to never fully turn off

MatthewKuKanich, 2024-03-27:
> > > It did it again. I left it active and it randomly turned off. The thing that says if it’s active turned off.
> > 
> > 
> > This definitely seems like a flipper issue. What is your Flippers sleep mode set to?
> 
> My sleep mode is 120s before the screen dims but it is set to never fully turn off

Is you sleep set to Legacy?

kenneth7398, 2024-03-27:
Sorry how do you check on momentum I forgot. I believe I have it set to the new one tho that is supposed to extend the battery life up to a month.

YupengLai4, 2024-04-18:
> Is you sleep set to Legacy?

I have the same issues on **Xtreme** _Default_ sleep method, Do you mean we need to change it to _Legacy_?

MatthewKuKanich, 2024-04-18:
> I have the same issues on **Xtreme** _Default_ sleep method, Do you mean we need to change it to _Legacy_?

No, quite the opposite. It should be working on _Default_. This seems to be an issue in the Xtreme firmware. Try flashing momentum

YupengLai4, 2024-04-18:
> This seems to be an issue in the Xtreme firmware.

It might be a problem, I always get out of memory when I open other app with FindMy in the background.

> Try flashing momentum

Wow, a new firmware that I never heard. Whats the different between Xtreme and momentum and which is better?

crazyquark, 2025-04-02:
I am on the latest Momentum FW and I have this issue. It started when I upgraded to 009, I think.