MatthewKuKanich/FindMyFlipper #37: Keys folder not being created

electrickid621, 2024-03-27:
Every other part of the setup process has gone smoothly, but after opening generate_keys.py, entering 1 as the number of keys to generate, and skipping the name entry, the window immediately closes, and no keys folder is made. Help!

electrickid621, 2024-03-27:
Nevermind. The directions need to be clearer. You have to run the script in the terminal, not just double-click on it.

closed 2024-03-27 by electrickid621

FatherDivine, 2024-03-27:
Did you contribute by forking the repo and proposing those "clearer directions" changes to the [README.md](https://github.com/MatthewKuKanich/FindMyFlipper/blob/main/AirTagGeneration/README.md) file?

electrickid621, 2024-03-27:
I'll do that now

MatthewKuKanich, 2024-03-27:
> Nevermind. The directions need to be clearer. You have to run the script in the terminal, not just double-click on it.

I made sure to mention this in the video tutorial and the instructions imply this since you need to `cd` into the directory of the python files. I did update the instructions to be more clear.

electrickid621, 2024-03-28:
I saw that afterwards, thanks. Sorry to bother you.