MatthewKuKanich/FindMyFlipper #39: ModuleNotFoundError: No module named 'Crypto'

francoiscamilleri84, 2024-03-28:
When running request_reports.py I am getting the below error:

/FindMyFlipper/AirTagGeneration/cores/pypush_gsa_icloud.py", line 15, in <module>
    from Crypto.Hash import SHA256
ModuleNotFoundError: No module named 'Crypto'

I have installed all requirements modules and Google so far has been unhelpful. Have tried installing cryptodome to try and remedy with no success either.

francoiscamilleri84, 2024-03-28:
> While inside the python virtual environment reinstall the module. Alternatively you can run the requirements.txt script with pip to install the dependencies

Interestingly, that's exactly the steps I followed. All requirements seem to have been met:
```

(venv) francoiscamilleri@Francoiss-MacBook-Pro AirTagGeneration % pip3 install -r requirements.txt
Requirement already satisfied: requests~=2.31.0 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 1)) (2.31.0)
Requirement already satisfied: urllib3~=2.1.0 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 2)) (2.1.0)
Requirement already satisfied: cryptography~=41.0.7 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 3)) (41.0.7)
Requirement already satisfied: pbkdf2~=1.3 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 4)) (1.3)
Requirement already satisfied: srp~=1.0.20 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 5)) (1.0.20)
Requirement already satisfied: fastapi~=0.104.1 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 6)) (0.104.1)
Requirement already satisfied: uvicorn~=0.24.0.post1 in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 7)) (0.24.0.post1)
Requirement already satisfied: folium in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 8)) (0.16.0)
Requirement already satisfied: pandas in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 9)) (2.2.1)
Requirement already satisfied: certifi in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 10)) (2024.2.2)
Requirement already satisfied: paho-mqtt in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 11)) (2.0.0)
Requirement already satisfied: pycryptodome in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 12)) (3.20.0)
Requirement already satisfied: python-multipart in ./venv/lib/python3.11/site-packages (from -r requirements.txt (line 13)) (0.0.9)
Requirement already satisfied: charset-normalizer<4,>=2 in ./venv/lib/python3.11/site-packages (from requests~=2.31.0->-r requirements.txt (line 1)) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.11/site-packages (from requests~=2.31.0->-r requirements.txt (line 1)) (3.6)
Requirement already satisfied: cffi>=1.12 in ./venv/lib/python3.11/site-packages (from cryptography~=41.0.7->-r requirements.txt (line 3)) (1.16.0)
Requirement already satisfied: six in ./venv/lib/python3.11/site-packages (from srp~=1.0.20->-r requirements.txt (line 5)) (1.16.0)
Requirement already satisfied: anyio<4.0.0,>=3.7.1 in ./venv/lib/python3.11/site-packages (from fastapi~=0.104.1->-r requirements.txt (line 6)) (3.7.1)
Requirement already satisfied: pydantic!=1.8,!=1.8.1,!=2.0.0,!=2.0.1,!=2.1.0,<3.0.0,>=1.7.4 in ./venv/lib/python3.11/site-packages (from fastapi~=0.104.1->-r requirements.txt (line 6)) (2.6.4)
Requirement already satisfied: starlette<0.28.0,>=0.27.0 in ./venv/lib/python3.11/site-packages (from fastapi~=0.104.1->-r requirements.txt (line 6)) (0.27.0)
Requirement already satisfied: typing-extensions>=4.8.0 in ./venv/lib/python3.11/site-packages (from fastapi~=0.104.1->-r requirements.txt (line 6)) (4.10.0)
Requirement already satisfied: click>=7.0 in ./venv/lib/python3.11/site-packages (from uvicorn~=0.24.0.post1->-r requirements.txt (line 7)) (8.1.7)
Requirement already satisfied: h11>=0.8 in ./venv/lib/python3.11/site-packages (from uvicorn~=0.24.0.post1->-r requirements.txt (line 7)) (0.14.0)
Requirement already satisfied: branca>=0.6.0 in ./venv/lib/python3.11/site-packages (from folium->-r requirements.txt (line 8)) (0.7.1)
Requirement already satisfied: jinja2>=2.9 in ./venv/lib/python3.11/site-packages (from folium->-r requirements.txt (line 8)) (3.1.3)
Requirement already satisfied: numpy in ./venv/lib/python3.11/site-packages (from folium->-r requirements.txt (line 8)) (1.26.4)
Requirement already satisfied: xyzservices in ./venv/lib/python3.11/site-packages (from folium->-r requirements.txt (line 8)) (2023.10.1)
Requirement already satisfied: python-dateutil>=2.8.2 in ./venv/lib/python3.11/site-packages (from pandas->-r requirements.txt (line 9)) (2.9.0.post0)
Requirement already satisfied: pytz>=2020.1 in ./venv/lib/python3.11/site-packages (from pandas->-r requirements.txt (line 9)) (2024.1)
Requirement already satisfied: tzdata>=2022.7 in ./venv/lib/python3.11/site-packages (from pandas->-r requirements.txt (line 9)) (2024.1)
Requirement already satisfied: sniffio>=1.1 in ./venv/lib/python3.11/site-packages (from anyio<4.0.0,>=3.7.1->fastapi~=0.104.1->-r requirements.txt (line 6)) (1.3.1)
Requirement already satisfied: pycparser in ./venv/lib/python3.11/site-packages (from cffi>=1.12->cryptography~=41.0.7->-r requirements.txt (line 3)) (2.21)
Requirement already satisfied: MarkupSafe>=2.0 in ./venv/lib/python3.11/site-packages (from jinja2>=2.9->folium->-r requirements.txt (line 8)) (2.1.5)
Requirement already satisfied: annotated-types>=0.4.0 in ./venv/lib/python3.11/site-packages (from pydantic!=1.8,!=1.8.1,!=2.0.0,!=2.0.1,!=2.1.0,<3.0.0,>=1.7.4->fastapi~=0.104.1->-r requirements.txt (line 6)) (0.6.0)
Requirement already satisfied: pydantic-core==2.16.3 in ./venv/lib/python3.11/site-packages (from pydantic!=1.8,!=1.8.1,!=2.0.0,!=2.0.1,!=2.1.0,<3.0.0,>=1.7.4->fastapi~=0.104.1->-r requirements.txt (line 6)) (2.16.3)

```

francoiscamilleri84, 2024-03-28:
Ok, cleared all files in venv - ran everything again. It now works. 

Thank you for your time

Neuropsihopuppu, 2024-04-02:
tried folowing all the steps listed here and it still does not work for me

francoiscamilleri84, 2024-04-04:
> tried folowing all the steps listed here and it still does not work for me
I went to the venv folder on my machine through finder on Mac (not via terminal) deleted everything from there and did a fresh install from requirements.txt and this solved the issue for me.

Neuropsihopuppu, 2024-04-04:
> > tried folowing all the steps listed here and it still does not work for me
> 
> I went to the venv folder on my machine through finder on Mac (not via terminal) deleted everything from there and did a fresh install from requirements.txt and this solved the issue for me.

I did the exact same thing,i even deleted absolutely everything and started from zero and redownloaded and reinstalled everything and i still get the same error :(

JackalopeZer0, 2024-08-13:
> I did the exact same thing,i even deleted absolutely everything and started from zero and redownloaded and reinstalled everything and i still get the same error :(

Same here, I am having the exact same issue.