MatthewKuKanich/FindMyFlipper #40: Flipper only pings when app is open

Salty-Coder, 2024-03-30:
On opening the FindMyFlipper app, I turn on broadcast.
It's location data suddenly becomes available on Find My as expected.
I then press the back button to exit the app, and it's location is no longer pinged to Find My.
Opening the app again, it shows "_Broadcast Inactive_" again.

There is no transmit icon on the home screen, as I have heard this is meant to be present, from reading another issue.

I am running `XFW-DEV 09-03-2024`.

MatthewKuKanich, 2024-03-30:
> On opening the FindMyFlipper app, I turn on broadcast.
> It's location data suddenly becomes available on Find My as expected.
> I then press the back button to exit the app, and it's location is no [...]

I see the issue! The back button exits the app and turns off broadcasting. That's as intended, to run in the background you need to press the left button on the d-pad. That's what the arrow is supposed to mean. I'll see if I can find a better icon. But yeah just press left and you're set

Salty-Coder, 2024-03-30:
Ah, I probably should have realised that.
Working now, thanks.

closed 2024-03-30 by Salty-Coder