MatthewKuKanich/FindMyFlipper #41: Unable to run anisette on Raspberry Pi 4 Model B Rev 1.5 (ARMv8)

Joaav, 2024-04-02:
Basic info:

```
Operating System: Raspbian GNU/Linux 11 (bullseye)
Kernel: Linux 6.1.21-v8+
Architecture: arm64
```

```
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: b03115
Serial		: 10000000e7d160c0
Model		: Raspberry Pi 4 Model B Rev 1.5
```

I am running command: `docker run -d --restart always --name anisette-v3 -p 6969:6969 dadoum/anisette-v3-server:latest`

After that container is running and returns following with input `docker container ls`
```
CONTAINER ID   IMAGE                              COMMAND                  CREATED          STATUS         PORTS                                       NAMES
8ed8f6f8f810   dadoum/anisette-v3-server:latest   "/opt/anisette-v3-se…"   58 seconds ago   Up 2 seconds   0.0.0.0:6969->6969/tcp, :::6969->6969/tcp   anisette-v3
```

Unfortunately input `curl https://localhost:6969` returns:
`curl: (7) Failed to connect to localhost port 6969: Connection refused`
or 
`curl: (56) Recv failure: Connection reset by peer`

When running `docker container ls` again I get no information on ports

```
CONTAINER ID   IMAGE                              COMMAND                  CREATED          STATUS                            PORTS     NAMES
8ed8f6f8f810   dadoum/anisette-v3-server:latest   "/opt/anisette-v3-se…"   10 minutes ago   Restarting (139) 36 seconds ago             anisette-v3
```

`docker logs` command doesn't give me anything

`docker service logs` command gives me this:

`Error response from daemon: This node is not a swarm manager. Use "docker swarm init" or "docker swarm join" to connect this node to swarm and try again.`

Any help would be appreciated as I am tearing my hair out and cannot get this to work

MatthewKuKanich, 2024-04-02:
Thanks for all the details! I actually just flashed a Pi 4 B just to run this server. I'll report back my experience with setting it up

Joaav, 2024-04-02:
> Thanks for all the details! I actually just flashed a Pi 4 B just to run this server. I'll report back my experience with setting it up

Thank you! Fingers crossed for getting it working on both mine and yours!

Joaav, 2024-04-15:
> Thanks for all the details! I actually just flashed a Pi 4 B just to run this server. I'll report back my experience with setting it up

Hi, any update on this? Does it work on yours Pi 4B?

MatthewKuKanich, 2024-04-18:
> > Thanks for all the details! I actually just flashed a Pi 4 B just to run this server. I'll report back my experience with setting it up
> 
> Hi, any update on this? Does it work on yours Pi 4B?

I've also not been successful at running the docker container on my Pi 4B. This seems to be an issue with container support for the Pi. Like mentioned above, this is an issue for the anisette docker image. Unfortunately, this is out of my scope. If I find a solution I will update here though.

Joaav, 2024-04-18:
> I've also not been successful at running the docker container on my Pi 4B. This seems to be an issue with container support for the Pi. Like mentioned above, this is an issue for the anisette docker [...]

Thank you! Its good to know it's not me. I posted there but still no response. If I will be able to get it working I will post here too.