MatthewKuKanich/FindMyFlipper #42: Suggestion: Add live battery meter

runestone74, 2024-04-03:
Hi, it seems like the battery in Apples "Find my" is stuck to 100%, I guess it is a quickfix until proper battery management is implemented. Thanks for a great app!

MatthewKuKanich, 2024-04-04:
It's stuck at 100% because this is what our payload is set to advertise. I'm working on making this dynamic so we can see the actual battery level. Stay tuned, coming soon!

nortakales, 2024-05-04:
As far as I can tell, Apple has removed the ability to see battery % in the Find My app. I don't see it for my Flipper or any of my other air tags. https://www.macrumors.com/2022/07/21/apple-removes-airtag-battery-level-icon-find-my/

MatthewKuKanich, 2024-05-04:
I have added this feature in the latest update! Battery reporting now reports the Flippers battery. There are only 4 battery states so the % in divided and then put into one of those states that matches the closest. Even with apple not showing the battery % in app, the data is still stored and parsed.

So you can see the working battery % with open haystack, macless haystack, my method, android, and any BLE scanner.

closed 2024-05-14 by MatthewKuKanich

runestone74, 2024-06-02:
Hi again, running v3.5 on Momentum, and it seems like the battery is only updated when going in to the app, not when using it in the background.