MatthewKuKanich/FindMyFlipper #43: Google Find My Device network has been rolled out. It would be great to integrate it.

MinecraftFuns, 2024-04-04:
According to [r/GooglePixel](https://www.reddit.com/r/GooglePixel/comments/1buws13/google_may_have_started_rolling_the_find_my/), it seems that the Google Find My Device network has been rolled out. It would be great to have my Flipper work with the Google network since I don't use any Apple devices.

MinecraftFuns, 2024-04-04:
I've searched the AOSP repositories and couldn't find anything related to the Find My Device network. There is a possibility that they decided to keep this part of the code proprietary, and reverse engineering the protocol may take some time. However, it certainly isn't impossible, as there are multiple tracking tag makers who have already implemented the protocol in their yet-to-be-released products.

MatthewKuKanich, 2024-04-04:
I'm definitely interested in bringing this functionality to the flipper. I'll be on the look for info

zacharyweiss, 2024-04-17:
@MatthewKuKanich here's Google's "Network Accessory Specification" for the Find My Device network: https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn

MrOscarvs, 2024-05-27:
This would be the ultimate addition, hope it gets added!

thedankeststorm, 2025-02-04:
Bumping this, wondering if this is feasible now

vguttmann, 2025-02-13:
There is been a feature-minimal working build for ESP32 devices now: https://github.com/leonboe1/GoogleFindMyTools

aaronjamt, 2025-02-13:
> There is been a feature-minimal working build for ESP32 devices now: https://github.com/leonboe1/GoogleFindMyTools

I've implemented this over on my fork: https://github.com/aaronjamt/FindMyFlipper

It seems to be working for me. As the warning in that repo says,it doesn't work with the web or mobile apps, only with the Python script, but that script does indeed give my current location. I'll go for a walk with my Flipper later and verify it records the location properly.

mhalano, 2025-03-07:
> > There is been a feature-minimal working build for ESP32 devices now: https://github.com/leonboe1/GoogleFindMyTools
> 
> I've implemented this over on my fork: https://github.com/aaronjamt/FindMyFlipper
> 
> It seems to be working for me. As the warning in that repo says,it doesn't work with the web or mobile apps, only with the Python script, but that script does indeed give my current location. I'll go for a walk with my Flipper later and verify it records the location properly. Edit: can confirm, it is definitely working.

@aaronjamt Could you please create a PR for your changes? I also would like to use this feature.

aaronjamt, 2025-03-07:
> @aaronjamt Could you please create a PR for your changes? I also would like to use this feature.

I was holding off until the key rotation algorithm had been sorted out, so that it's able to be shown in the official Find My Devices app, rather than having to use a command-line tool to locate it. I guess I'll leave that decision up to @MatthewKuKanich

MatthewKuKanich, 2025-03-07:
I'd be happy to take a look at a PR! I have some ideas on a key rotation workaround but I need to do some more testing

aaronjamt, 2025-03-07:
> I'd be happy to take a look at a PR! I have some ideas on a key rotation workaround but I need to do some more testing

Sounds good, I created #115.

jbaenaxd, 2025-12-30:
> https://www.reddit.com/r/flipperzero/comments/1odx9hh/comment/nkzgjjd/?context=3

News about this? Got it released already?

aaronjamt, 2025-12-30:
> > https://www.reddit.com/r/flipperzero/comments/1odx9hh/comment/nkzgjjd/?context=3
> 
> News about this? Got it released already?
> 
> 

It's been added, see #115