MatthewKuKanich/FindMyFlipper #44: Always falls back to Apple Network.

tkbstudios, 2024-04-04:
When I launch FindMy, register my tag generated using the tutorial you put on youtube, select Samsung Network, etc..

When I get back to the main screen and press the left arrow to let it run in the background, when I re-open the app, it shows me Apple Network.

How do I fix this?

MatthewKuKanich, 2024-04-04:
> When I launch FindMy, register my tag generated using the tutorial you put on youtube, select Samsung Network, etc..
> 
> When I get back to the main screen and press the left arrow to let it run in the background, when I re-open the app, it shows me Apple Network.
> 
> How do I fix this?

Thanks for letting me know, I'll look into it! Make sure you're running the latest version as I've changed the network switching logic to make it more defined. The type of tag you register gets saved to a settings file on your Flipper. Upon opening the app it will load everything based off of that setting file (Mac, payload, tag type, etc).

MatthewKuKanich, 2024-04-04:
Didn't mean to close it 😅

tkbstudios, 2024-04-04:
No problem! I'll look forward to your response.

findmy_state.txt
```
Filetype: FindMy Flipper State
Version: 1
beacon_active: false
broadcast_interval: 4
transmit_power: 6
tag_type: 0
show_mac: false
mac: ****
data: ****
```

MatthewKuKanich, 2024-04-04:
> No problem! I'll look forward to your response.
> 
> findmy_state.txt
> ```
> Filetype: FindMy Flipper State
> Version: 1
> beacon_active: false
> broadcast_interval: 4
> transmit_power: 6
> tag_type: 0
> show_mac: false
> mac: ****
> data: ****
> ```

Thanks for sharing the file, this helps a lot. Tag_type 0 indicates airtag, I'm going to make sure that updates when clicking samsung

tkbstudios, 2024-04-05:
Thanks! I'm looking forward for a new release :)