MatthewKuKanich/FindMyFlipper #48: I couldn't imput 2FA code

Itou-Kouki, 2024-04-06:
I run `request_reports.py`, it stops with an error before imput F2A.
```
pyprovision is not installed, querying http://localhost:6969 for an anisette server
C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\urllib3\connectionpool.py:1099: InsecureRequestWarning: Unverified HTTPS request is being made to host 'gsa.apple.com'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#tls-warnings
  warnings.warn(
pyprovision is not installed, querying http://localhost:6969 for an anisette server
C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\urllib3\connectionpool.py:1099: InsecureRequestWarning: Unverified HTTPS request is being made to host 'gsa.apple.com'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#tls-warnings
  warnings.warn(
2FA required, requesting code
pyprovision is not installed, querying http://localhost:6969 for an anisette server
C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\urllib3\connectionpool.py:1099: InsecureRequestWarning: Unverified HTTPS request is being made to host 'gsa.apple.com'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#tls-warnings
  warnings.warn(
'latin-1' codec can't encode characters in position 0-1: ordinal not in range(256)
```

I think cause of email adress.
I use Gmail adress on AppleID.
Does someone know how to solve it?

EricYinhy, 2024-04-17:
[image]
I have the same problem

Hari-B6, 2024-07-16:
Your Windows language has to be set to English. That's what worked for me.

Itou-Kouki, 2024-07-16:
I wrote something similar. https://github.com/MatthewKuKanich/FindMyFlipper/issues/59#issuecomment-2072246995

closed 2024-07-16 by Itou-Kouki