MatthewKuKanich/FindMyFlipper #59: Error on request_reports.py

EricYinhy, 2024-04-17:
[image]

'latin-1' codec can't encode characters in position 0-5: ordinal not in range(256)

EricYinhy, 2024-04-17:
It's the same problem on this ↓
Does anyone know how to solve it?
https://github.com/MatthewKuKanich/FindMyFlipper/issues/48

MatthewKuKanich, 2024-04-18:
Interesting. This seems to be an issue with ascii characters and likely keyboard layout issues. What is your keyboard language?

EricYinhy, 2024-04-18:
> Interesting. This seems to be an issue with ascii characters and likely keyboard layout issues. What is your keyboard language?

English US,I've tried to switch keyboard language,but it doesn't work.

Itou-Kouki, 2024-04-18:
> Interesting. This seems to be an issue with ascii characters and likely keyboard layout issues. What is your keyboard language?

I use JIS keyboard but imput English at that time.

0ldF0x, 2024-04-21:
If you switch windows language to English US, works fine.

Itou-Kouki, 2024-04-23:
> If you switch windows language to English US, works fine.
Set the encoding in script to UTF8

Certainly,Language change to English to disaprear 'latin-1' codec error.