MatthewKuKanich/FindMyFlipper #65: RequestReport&Map doesn't work unless pandas and folium are installed system-wide

LoudSoftware, 2024-04-22:
https://github.com/MatthewKuKanich/FindMyFlipper/blob/367e4ae73f5fe6935be4c13b2d21f5ddb8977bf9/AirTagGeneration/RequestReport%26Map.py#L153

For some reason, the subprocess.run doesn't seem to respect the current environment, defaulting to the system python interpreter. If pandas and folium are not present outside of the venv, then the following will happen when calling the RequestReport&Map script:
```
[...]
{'lat': x, 'lon': x, 'conf': x, 'status': 0, 'timestamp': x, 'isodatetime': x', 'key': 'x', 'goog': 'x'}
found:   ['x']
missing: []
Reports have been successfully retrieved.
Data has been successfully exported to 'data.json'.
Script encountered an error.
Error: Traceback (most recent call last):
  File "C:\Users\x\FindMyFlipper\AirTagGeneration\advanced_map_loc.py", line 2, in <module>
    import pandas as pd
ModuleNotFoundError: No module named 'pandas'
```

MatthewKuKanich, 2024-04-23:
This is strange. I haven't seen this before, can you confirm that your virtual environment is active when running the script?

LoudSoftware, 2024-04-23:
I can confirm I'm in the venv when running the script. 
Stranger still, this only seems to be an issue on windows. I had no trouble on linux

MatthewKuKanich, 2024-04-23:
> I can confirm I'm in the venv when running the script. Stranger still, this only seems to be an issue on windows. I had no trouble on linux

That is def weird. I can't seem to recreate the issue on my Windows machine. This may be something specific to your Windows pc, but I'll leave it open and still be on the lookout for answers/others with this issue.

user890104, 2024-04-29:
It happens for me, too. Ubuntu 20.04.6 LTS with Python 3.9.5.

You can work around this by calling advanced_map_loc.py manually within the venv.
E.g. `./venv/bin/python advanced_map_loc.py`

super06, 2024-05-04:
make sure you have permission, that was my problem. so i gave it with chmod +x "file name"