MatthewKuKanich/FindMyFlipper #67: Cannot add generated keys to the iPhone

krtek, 2024-04-25:
# Steps to reproduce

1. Generate key file from scratch.
2. Import it to the Flipper and set Apple network.
3. Start Broadcast.
4. Open Find my app on the iPhone and try to add new AirTag.
5. Nothing happens. 
[image]
[image]

otisdog8, 2024-04-30:
I think this device presents itself as an airtag that is already owned by someone, and it sends the pings of its location to apple, as opposed to a "real" airtag that you can enroll to a device.

krtek, 2024-04-30:
Reading the README.md again and you're right! Thank you!

closed 2024-04-30 by krtek