MatthewKuKanich/FindMyFlipper #79: KeyError: 'isodatetime'

lewilou22, 2024-06-19:
Works within a few days of setup without issues.  after a few days and over 400 reports, the code will give this error:

```(venv) C:\FindMyFlipper\AirTagGeneration>python RequestReportMap.py
pyprovision is not installed, querying http://localhost:6969 for an anisette server
200: 486 reports received.
0 reports used.
found:   []
missing: ['e8a3fcdbda86']
Reports have been successfully retrieved.
Data has been successfully exported to 'data.json'.
Script encountered an error.
Error: Traceback (most recent call last):
  File "C:\FindMyFlipper\AirTagGeneration\advanced_map_loc.py", line 23, in <module>
    df['datetime'] = pd.to_datetime(df['isodatetime'])
  File "C:\Users\choco\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\core\frame.py", line 4102, in __getitem__
    indexer = self.columns.get_loc(key)
  File "C:\Users\choco\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\core\indexes\range.py", line 417, in get_loc
    raise KeyError(key)
KeyError: 'isodatetime'```

morty29, 2024-06-19:
I have the same problem. It looks like the problem is on the flipper side. The app stops broadcasting 2 days after activation and doesn't want to start again even if you flip the activation switch on and off. What helps to restore the broadcasting is reimporting the same key. After that data appears on the server and map is generated, but obviously starting from the time when key was reimported.

MatthewKuKanich, 2024-06-19:
> I have the same problem. It looks like the problem is on the flipper side. The app stops broadcasting 2 days after activation and doesn't want to start again even if you flip the activation switch [...]

This is very odd. Especially since there shouldn't be any changes between an imported key and the broadcast key aside from the battery bytes. Maybe the battery updates break it somehow, I'll need to do some testing. Try charging your Flipper to 100% next time it stops broadcasting and see if that fixes it.

morty29, 2024-06-23:
My flipper is now below the charge level that it was when I encountered the issue, and a longer period of time went through and beacon is broadcasting just fine.  But one time checking it I saw it stopped and realised that it stopped exactly at the moment I was checking it previously. So my main theory right now is that it works fine, its just me habitually exiting the app with a back button rather than a left nav key without noticing it.
And impression of it not reacting for turning on and off was likely from a delay of a round trip through the whole apple infrastructure. 
I am not OP, but from my side everything looks fine and the problem seem to be between the computer and the chair) So I am fine with closing the issue.

Ayushskull7, 2025-04-05:
Hello, I am currently facing the same issue. My Flipper is charged to approximately 70%, yet I receive this error message. I am certain that I have not exited the app by clicking the back button, and it indicates "Broadcast Active" every time I open the app. I have also tried re-importing keys, but the issue persists. Could you please explain how to get past this issue? I can provide logs if necessary. Thanks

```
(venv) E:\FindMyFlipper\AirTagGeneration>python "RequestReport&Map.py"
pyprovision is not installed, querying http://localhost:6969 for an anisette server
200: 1767 reports received.
0 reports used.
found:   []
missing: ['MAC ADDRESS']
Reports have been successfully retrieved.
Data has been successfully exported to 'data.json'.
Script encountered an error.
Error: Traceback (most recent call last):
  File "E:\FindMyFlipper\AirTagGeneration\advanced_map_loc.py", line 23, in <module>
    df['datetime'] = pd.to_datetime(df['isodatetime'])
                                    ~~^^^^^^^^^^^^^^^
  File "C:\Users\ayush\AppData\Local\Programs\Python\Python313\Lib\site-packages\pandas\core\frame.py", line 4102, in __getitem__
    indexer = self.columns.get_loc(key)
  File "C:\Users\ayush\AppData\Local\Programs\Python\Python313\Lib\site-packages\pandas\core\indexes\range.py", line 417, in get_loc
    raise KeyError(key)
KeyError: 'isodatetime'
```