MatthewKuKanich/FindMyFlipper #80: issue with request_reports

CrypticxTetra, 2024-06-20:


lkjlkjlkj2012, 2024-08-08:
I has same question.

lkjlkjlkj2012, 2024-08-08:
I has same problem.

1e1, 2024-09-19:
I had the same issue. 
Reading the code, I see: https://github.com/MatthewKuKanich/FindMyFlipper/blob/main/AirTagGeneration/request_reports.py#L61
Running `python3 request_reports.py --regen --trusteddevice` works for me. 
By default Apple should send an SMS. As it's more expensive than a "push notification", there may be more controls.

PawiX25, 2024-09-26:
I'm getting a `'NoneType' object is not subscriptable` error and repeated warnings about unverified HTTPS requests to `gsa.apple.com`. The script also tries to query a local anisette server due to `pyprovision` not being installed. Running `python3 request_reports.py --regen --trusteddevice` doesn't resolve the issue either. Any advice?