MatthewKuKanich/FindMyFlipper #82: FindMyFlipper not updating location

Ixolaro, 2024-07-02:
Hi, im running Momentum Release firmware (Latest) and i generated an airtag key and i got only 10 reports from 2 hours ago (Not updating) with my iphone besides it. FindMyFlipper running in the background with 5-second interval and max DBm power. Yes, i did use the correct key and yes bluetooth is enabled on both devices.

Ixolaro, 2024-07-02:
nvm it updated... for some weird reason it sent location after 2 and a half hour...

zzzNTG, 2024-07-22:
I've experienced similar issues. Updates instantly when ran manually which indicates this could be a real bug. I'm running stock firmware.

zzzNTG, 2024-07-22:
I may have figured out my issue. When opening the FindMy app on the device you need to press the left arrow to get it to run in the background. When it is running in the background. You'll see a beacon icon in the left corner of the home screen (at least in stock firmware). 

Now I need to figure out how to run this on boot since now if the devices is rebooted FindMy will need to be manually started which is kind of annoying.