MatthewKuKanich/FindMyFlipper #89: Flipper isn't near you

lucidnx, 2024-07-30:
Soon after I configured app (cloned SmartTag), I am getting frequent notifications that tag is not near me. Is this WIP or this behavior is not expected? It seems to be happening while screen is off (device locked in my case), but can not confirm, just testing right now.

Interval 3s, 6dBm
BT ON

MatthewKuKanich, 2024-07-30:
This isn't expected behavior. It may have to do with your device locking. Try removing the lock/pin and see if that fixes it. If so I'll look into a workaround so you can still use pin

lucidnx, 2024-07-30:
I have tried to keep it alive (screen on). Issue still here.

lucidnx, 2024-07-30:
Hello! It may be some power saving option/function. Over night it stopped pinging completely. Only re-opening app worked. App was running tho, icon in statusbar was present. I am using Momentum FW

lucidnx, 2024-08-01:
Closer info:
1. I am using function to tell me when tag is not with me
2. I am getting notification that tag is not near me maybe every 5 minutes
3. using Momentum CFW, Samsung's SmartTag (v1)
4. Lock on flipper does not change anything

Is there any chance that I'v got something wrong while reading payload?
Maybe from time to time, ping is not being sent/received on phone and it will immedialy throw notification. I will check that with SDR if it's really needed. Also it may be something else, not ping itself, but I dunno what. Somebody else with SmartTag should help to reproduce this issue.

Otherwise seems that it's working, just it's kinda annoying if that notification is turned on or it will never tell me if I leave it somewhere and go away.