MatthewKuKanich/FindMyFlipper #108: AirTags no longer work with iOS 18.2

nortakales, 2024-12-18:
The only confirmation I have is that my iPhone on 18.2 is no longer able to see my Flipper Zero. But my iPad, which is not yet updated, is able to see it. Also, at the same time I updated, I started getting the "unknown air tag is following you" warning, which I am assuming to be my Flipper Zero.

CytoStorm, 2024-12-25:
I’m having the same issue. Can anyone else confirm this? Both my iPad and iPhone are on 18.2. I guess I need to try my Mac as well.

porpup, 2024-12-26:
I had same problem. I founded work around: I shared paired AirTag from iPhone with firmware 18.2 with another iPhone with firmware 18.1.1 (with different account). Than cloned that AirTag to Flipper Zero. After it was recognized by iPhone with 18.1.1 firmware it started to be recognized by iPhone with 18.2 firmware. After stopping sharing it with iPhone with 18.1.1 firmware it worked as it supposed to on iPhone with 18.2 firmware. Hope this worked not just for me.

agentile27, 2025-01-14:
I haven't been able to get this work on my devices, all running 18.2. I don't have anything running an older version of iOS to test this with.

thaycafe, 2025-02-04:
> I haven't been able to get this work on my devices, all running 18.2. I don't have anything running an older version of iOS to test this with.

same here, waiting for a solution.

rodgomesc, 2025-02-07:
I had a similar problem. After the whole configuration process, cloning the AirTag wasn't being updated in the "Find My" app on iOS 18.3. It turns out that just opening "Find My" from macOS shows the Flipper, and now going back to iOS is working there as well. I'm not sure how it was solved, but it just worked.

CafardNRV, 2025-05-23:
Same here unable to see the AirTag in the FindMy app

frozenfoxx, 2025-09-04:
Same issue here