MatthewKuKanich/FindMyFlipper #110: key generation fail

vextr2009, 2024-12-26:
hi so when i open generate_keys.py it opens terminal then it asks how many keys do i want i type 1 then there is a name i skip it by pressing enter but after that the cmd just shuts away and no folder is generated or any key i have everything instaled corectly and all the steps up to this are fine can anyone help i tried restarting everithing instal/uninstal nothing worked

thaycafe, 2025-01-14:
Can you add some debug print into generate_keys.py?

This will help us find out exactly when the code crashed.

Example:
[image]

vextr2009, 2025-01-14:
Yeah thanks i will try

the-one-with-raspberry, 2025-08-31:
The problem is at line 73, in the `hashes.Hash` method. Modern versions of `cryptography` require a `backend` parameter. This can be solved by adding `backend=default_backend()` to the method's args.